Interbotix/interbotix_ros_manipulators #236: [Question]: Problem with ROS Standard Software Setup

abdullah594, 2024-12-22:
### Question

Hello, 

I'm trying to go to do the ROS Standard Software Setup for my PX-100 robotic arm. I ran all the lines of codes in the guideline provided in the link below: 
https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/software_setup.html#remote-install

However, when I reached the "[Installation Checks]" part of the guideline, I had two problems:  

1- The [udev Rules] checker didn't get me the expected output (i.e. ttyDXL). Nothing returned when I typed
```shell
$ls /dev | grep ttyDXL
``` 
I should get an output similar to: 
```shell
ttyDXL
``` 

2- When I try to do the [Interbotix ROS Packages] installation checker by typing the following: 
```shell
$ source /opt/ros/noetic/setup.bash
$ source ~/interbotix_ws/devel/setup.bash
$ rospack list | grep interbotix
``` 
I should get a bunch of packages. I got some of them, but some of them are missing. What I got installed are the following: 
```shell
interbotix_xsarm_control /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_control
interbotix_xsarm_descriptions /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_descriptions
interbotix_xsarm_diagnostic_tool /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_diagnostic_tool
interbotix_xsarm_dual /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_dual
interbotix_xsarm_dual_joy /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_dual_joy
interbotix_xsarm_gazebo /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_gazebo
interbotix_xsarm_joy /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_joy
interbotix_xsarm_moveit /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_ros_xsarms_dual_moveit
interbotix_xsarm_moveit_interface /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_perception
interbotix_xsarm_pid /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_pid
interbotix_xsarm_puppet /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_puppet
interbotix_xsarm_ros_control /home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_ros_control
``` 
and the following packages are missing 
```shell
interbotix_common_modules /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_common_modules
interbotix_landmark_modules /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_landmark_modules
interbotix_moveit_interface /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_moveit_interface
interbotix_perception_modules /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_perception_toolbox/interbotix_perception_modules
interbotix_tf_tools /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_common_toolbox/interbotix_tf_tools
interbotix_xs_modules /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_xs_toolbox/interbotix_xs_modules
interbotix_xs_msgs /home/$USER/interbotix_ws/src/interbotix_ros_core/interbotix_ros_xseries/interbotix_xs_msgs
interbotix_xs_ros_control /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_xs_toolbox/interbotix_xs_ros_control
interbotix_xs_rviz /home/$USER/interbotix_ws/src/interbotix_ros_toolboxes/interbotix_xs_toolbox/interbotix_xs_rviz
interbotix_xs_sdk /home/$USER/interbotix_ws/src/interbotix_ros_core/interbotix_ros_xseries/interbotix_xs_sdk
/home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/examples/interbotix_xsarm_moveit_interface
interbotix_xsarm_perception 
/home/$USER/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_moveit
interbotix_xsarm_moveit_dual 
``` 

Because these packages are missing, I'm facing some error messages when I try to go over the ROS 1 Quickstart Guide:
https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/quickstart.html

### Robot Model

px100

### Operating System

Ubuntu 20.04

### ROS Version

ROS 1 Noetic

lukeschmitt-tr, 2024-12-30:
> ```
> [ WARN] [1734881768.784773200]: Shutdown request received.
> [ WARN] [1734881768.814792920]: Reason given for shutdown: [[/px100/robot_state_publisher] Reason: new node registered with same name]
> ```

These warnings indicate that you are running multiple roslaunch sessions that run the same nodes without first terminating old sessions. Please shut down old roslaunch session before starting new ones.

> ```
> [ERROR] [1734880545.057303145]: [xs_sdk] Failed syncRead: [TxRxResult] Incorrect status packet!
> [ERROR] [1734880545.058282441]: [xs_sdk] Failed getSyncReadData for Present_Current: groupSyncRead getdata failed
> [ERROR] [1734880545.058381527]: [xs_sdk] Failed getSyncReadData for Present_Velocity: groupSyncRead getdata failed
> [ERROR] [1734880545.058396597]: [xs_sdk] Failed getSyncReadData for Present_Position: groupSyncRead getdata failed
> ```

Please see the troubleshooting steps mentioned here: https://docs.trossenrobotics.com/interbotix_xsarms_docs/troubleshooting.html#incorrect-status-packet

Please also check that your arm is powered with the provided 12V power supply.

lukeschmitt-tr, 2025-02-11:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2025-02-11 by lukeschmitt-tr