Interbotix/interbotix_ros_manipulators #239: [Bug]: WidowX gripper moving in y-direction opposite to the commanded input

abadithela, 2025-01-05:
### What happened?

Hi,

I am using the WidowX 250s on UBuntu 22.04 with ROS Noetic. I have two questions. First, I am using a policy to command changes in gripper position, and I am using the bot.arm.set_ee_cartesian_trajectory(x=dx, y=dy, z=dz, roll=droll, pitch=dpitch, yaw=dyaw) to update the gripper position. We're noticing the following failure mode about once or twice every 10 trials. The gripper moves in a dy position opposite to the commanded dy. For example, the commanded dy position can take a small positive value (which is supposed to shift the gripper to the left), and followed by a larger negative value (which is supposed to shift it to the right) as given below. However, what we observe is that it does not properly execute the commands of step 22, and continues shifting further to the left.

```
Running Step: 20

dx: -0.0025309075135737658
dy: 0.0035765364300459623
dz: 0.0039052797947078943
droll: 0.03742938116192818
dpitch: -0.002348142210394144
dyaw: -0.010451313108205795
Gripper: -0.002302885986864567

dx: -0.0013802630128338933
dy: 0.0037133574951440096
dz: 0.0026718107983469963
droll: 0.019720030948519707
dpitch: 0.002738431328907609
dyaw: -0.007126897107809782
Gripper: 0.002539239125326276

Running Step: 22

dx: 0.0013360296143218875
dy: -0.015529503114521503
dz: 0.011899501085281372
droll: -0.010677888989448547
dpitch: -0.006024511530995369
dyaw: 0.02880163863301277
Gripper: 0.014854628592729568

dx: 0.022568529471755028
dy: -0.03731013089418411
dz: 0.019864581525325775
droll: -0.05908364802598953
dpitch: -0.021504143252968788
dyaw: 0.08463582396507263
Gripper: -0.008733903057873249
```

The second question is regarding doing a Dynamixel firmware update. On the dynamixel wizard, all motors have an asterisk next to them, which I understand to mean that a firmware update is available. Would doing a firmware update resolve the above issue? More importantly, is the latest firmware update compatible with the ROS software of interbotix?

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 1 Noetic

lukeschmitt-tr, 2025-01-06:
Please [enable debug logging](https://github.com/Interbotix/interbotix_ros_manipulators/blob/a4b7e51150d604e2f16451b703fa5f324dd52520/interbotix_ros_xsarms/interbotix_xsarm_control/launch/xsarm_control.launch.py#L101) on the arm's xs_sdk provide the output when running your script.

Please also ensure that the commands you are providing to the arm do not exceed the joint limits and are feasible according to the solver used by the set_ee_cartesian_trajectory method.

abadithela, 2025-01-06:
What is the command to enable debug logging? We ran the following but it did not seem to do anything:

```
roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s xs_driver_logging_level:=debug
```

Yes, the commands we are providing to the arm do not exceed joint limits. Every argument to set_ee_cartesian_trajectory is on the order of 0.01 at most.

abadithela, 2025-01-07:
When I've tried debug logging, it gives me the following output on terminal when I'm seeing this failure:

```
[DEBUG] [1736289088.502407052]: [xs_sdk::robot_set_motor_registers] ID: 7, writing reg: Profile_Acceleration, value: 300.
[DEBUG] [1736289088.513375646]: [xs_sdk::robot_set_motor_registers] ID: 8, writing reg: Profile_Acceleration, value: 300.
[DEBUG] [1736289088.524488435]: Connection::drop(2)
[DEBUG] [1736289088.524571491]: TCP socket [17] closed
[DEBUG] [1736289088.524605756]: Connection::drop(0)
[DEBUG] [1736289088.524642868]: Connection::drop(2)
[DEBUG] [1736289088.524674965]: Connection::drop(2)
[DEBUG] [1736289088.527426722]: Time jumped forward by [0.052035] for timer of period [0.010000], resetting timer (current=1736289088.527397, next_expected=1736289088.475362)
[DEBUG] [1736289088.527668775]: Accepted connection on socket [7], new socket [17]
[DEBUG] [1736289088.528016503]: Adding tcp socket [17] to pollset
[DEBUG] [1736289088.528081768]: TCPROS received a connection from [128.112.36.33:48132]
[DEBUG] [1736289088.528143624]: Async socket[17] is connected
[DEBUG] [1736289088.528197920]: Connection: Creating ServiceClientLink for service [/wx250s/get_robot_info] connected to [callerid=[/wx250s_robot_manipulation] address=[TCPROS connection on port 57753 to [128.112.36.33:48132 on socket 17]]]
[DEBUG] [1736289088.528236881]: Service client [/wx250s_robot_manipulation] wants service [/wx250s/get_robot_info] with md5sum [873f2223b25122251d4d3641d8411cf6]
[DEBUG] [1736289088.532888031]: Parsing robot urdf xml string
[DEBUG] [1736289088.533912279]: successfully added a new material 'interbotix_black'
[DEBUG] [1736289088.533947897]: setting link 'world' material
[DEBUG] [1736289088.533969381]: successfully added a new link 'world'
[DEBUG] [1736289088.534101097]: setting link 'wx250s/base_link' material
[DEBUG] [1736289088.534131636]: setting link 'wx250s/base_link' material to 'interbotix_black'
[DEBUG] [1736289088.534161004]: setting link 'wx250s/base_link' material to 'interbotix_black'
[DEBUG] [1736289088.534192632]: successfully added a new link 'wx250s/base_link'
[DEBUG] [1736289088.534309757]: setting link 'wx250s/shoulder_link' material
[DEBUG] [1736289088.534340933]: setting link 'wx250s/shoulder_link' material to 'interbotix_black'
[DEBUG] [1736289088.534372364]: setting link 'wx250s/shoulder_link' material to 'interbotix_black'
[DEBUG] [1736289088.534401759]: successfully added a new link 'wx250s/shoulder_link'
[DEBUG] [1736289088.534514846]: setting link 'wx250s/upper_arm_link' material
[DEBUG] [1736289088.534545433]: setting link 'wx250s/upper_arm_link' material to 'interbotix_black'
[DEBUG] [1736289088.534575510]: setting link 'wx250s/upper_arm_link' material to 'interbotix_black'
[DEBUG] [1736289088.534605151]: successfully added a new link 'wx250s/upper_arm_link'
[DEBUG] [1736289088.534713001]: setting link 'wx250s/upper_forearm_link' material
[DEBUG] [1736289088.534744027]: setting link 'wx250s/upper_forearm_link' material to 'interbotix_black'
[DEBUG] [1736289088.534774274]: setting link 'wx250s/upper_forearm_link' material to 'interbotix_black'
[DEBUG] [1736289088.534803928]: successfully added a new link 'wx250s/upper_forearm_link'
[DEBUG] [1736289088.534917263]: setting link 'wx250s/lower_forearm_link' material
[DEBUG] [1736289088.534946454]: setting link 'wx250s/lower_forearm_link' material to 'interbotix_black'
[DEBUG] [1736289088.534976272]: setting link 'wx250s/lower_forearm_link' material to 'interbotix_black'
[DEBUG] [1736289088.535005619]: successfully added a new link 'wx250s/lower_forearm_link'
[DEBUG] [1736289088.535122201]: setting link 'wx250s/wrist_link' material
[DEBUG] [1736289088.535151825]: setting link 'wx250s/wrist_link' material to 'interbotix_black'
[DEBUG] [1736289088.535181944]: setting link 'wx250s/wrist_link' material to 'interbotix_black'
[DEBUG] [1736289088.535209313]: successfully added a new link 'wx250s/wrist_link'
[DEBUG] [1736289088.535320013]: setting link 'wx250s/gripper_link' material
[DEBUG] [1736289088.535351130]: setting link 'wx250s/gripper_link' material to 'interbotix_black'
[DEBUG] [1736289088.535379534]: setting link 'wx250s/gripper_link' material to 'interbotix_black'
[DEBUG] [1736289088.535409606]: successfully added a new link 'wx250s/gripper_link'
[DEBUG] [1736289088.535456848]: setting link 'wx250s/ee_arm_link' material
[DEBUG] [1736289088.535486272]: successfully added a new link 'wx250s/ee_arm_link'
[DEBUG] [1736289088.535602111]: setting link 'wx250s/gripper_prop_link' material
[DEBUG] [1736289088.535634199]: setting link 'wx250s/gripper_prop_link' material to 'interbotix_black'
[DEBUG] [1736289088.535661157]: setting link 'wx250s/gripper_prop_link' material to 'interbotix_black'
[DEBUG] [1736289088.535691472]: successfully added a new link 'wx250s/gripper_prop_link'
[DEBUG] [1736289088.535805887]: setting link 'wx250s/gripper_bar_link' material
[DEBUG] [1736289088.535834631]: setting link 'wx250s/gripper_bar_link' material to 'interbotix_black'
[DEBUG] [1736289088.535861725]: setting link 'wx250s/gripper_bar_link' material to 'interbotix_black'
[DEBUG] [1736289088.535891422]: successfully added a new link 'wx250s/gripper_bar_link'
[DEBUG] [1736289088.535938454]: setting link 'wx250s/fingers_link' material
[DEBUG] [1736289088.535968763]: successfully added a new link 'wx250s/fingers_link'
[DEBUG] [1736289088.536088282]: setting link 'wx250s/left_finger_link' material
[... cut ...]
[DEBUG] [1736289091.652375084]: Received update for topic [/wx250s/commands/joint_trajectory] (0 publishers)
[DEBUG] [1736289091.652410981]: Publisher update for [/wx250s/commands/joint_trajectory]:  already have these connections: http://128.112.36.33:39009/, 
[DEBUG] [1736289091.652437254]: Disconnecting from publisher [/wx250s_robot_manipulation] of topic [/wx250s/commands/joint_trajectory] at [http://128.112.36.33:39009/]
[DEBUG] [1736289091.652465209]: Connection::drop(2)
[DEBUG] [1736289091.652495330]: Connection::drop(2)
[DEBUG] [1736289091.652522640]: Connection::drop(2)
```

abadithela, 2025-01-07:
For reference, here is the debug console output when I successfully command the robot to a certain state. I am curious about the `[DEBUG] [1736289534.655163769]: Socket -1 closed with (ERR|HUP|NVAL) events 8193: Unknown error -1`

```
[DEBUG] [1736289532.641370468]: [xs_sdk::robot_write_commands] ID: 1, writing position command 2048.
[DEBUG] [1736289532.641426540]: [xs_sdk::robot_write_commands] ID: 2, writing position command 874.
[DEBUG] [1736289532.641470506]: [xs_sdk::robot_write_commands] ID: 4, writing position command 3058.
[DEBUG] [1736289532.641515501]: [xs_sdk::robot_write_commands] ID: 6, writing position command 2048.
[DEBUG] [1736289532.641566570]: [xs_sdk::robot_write_commands] ID: 7, writing position command 2569.
[DEBUG] [1736289532.641617483]: [xs_sdk::robot_write_commands] ID: 8, writing position command 2048.
[DEBUG] [1736289534.654694451]: Socket [18] received 0/4 bytes, closing
[DEBUG] [1736289534.654874919]: TCP socket [18] closed
[DEBUG] [1736289534.654918014]: Connection::drop(0)
[DEBUG] [1736289534.654984207]: Connection to publisher [TCPROS connection on port 50784 to [128.112.36.33:45347 on socket 18]] to topic [/wx250s/commands/joint_group] dropped
[DEBUG] [1736289534.655126034]: getsockopt failed on socket [-1]
[DEBUG] [1736289534.655163769]: Socket -1 closed with (ERR|HUP|NVAL) events 8193: Unknown error -1
[DEBUG] [1736289534.655248454]: Socket [19] received 0/4 bytes, closing
[DEBUG] [1736289534.655335641]: TCP socket [19] closed
[DEBUG] [1736289534.655365889]: Connection::drop(0)
[DEBUG] [1736289534.655402802]: Connection to publisher [TCPROS connection on port 50794 to [128.112.36.33:45347 on socket 19]] to topic [/wx250s/commands/joint_single] dropped
[DEBUG] [1736289534.655461614]: getsockopt failed on socket [-1]
[DEBUG] [1736289534.655494054]: Socket -1 closed with (ERR|HUP|NVAL) events 8193: Unknown error -1
[DEBUG] [1736289534.655527065]: Socket [20] received 0/4 bytes, closing
[DEBUG] [1736289534.655599438]: TCP socket [20] closed
[DEBUG] [1736289534.655629903]: Connection::drop(0)
[DEBUG] [1736289534.655666590]: Connection to publisher [TCPROS connection on port 50800 to [128.112.36.33:45347 on socket 20]] to topic [/wx250s/commands/joint_trajectory] dropped
[DEBUG] [1736289534.655706785]: getsockopt failed on socket [-1]
[DEBUG] [1736289534.655737677]: Socket -1 closed with (ERR|HUP|NVAL) events 8193: Unknown error -1
[DEBUG] [1736289534.655947660]: Socket 11 closed with (ERR|HUP|NVAL) events 8192: Success
[DEBUG] [1736289534.656025198]: TCP socket [11] closed
[DEBUG] [1736289534.656055320]: Connection::drop(0)
[DEBUG] [1736289534.656099734]: Connection to subscriber [callerid=[/wx250s_robot_manipulation] address=[TCPROS connection on port 58241 to [128.112.36.33:53128 on socket 11]]] to topic [/wx250s/joint_states] dropped
[DEBUG] [1736289534.656137182]: Connection::drop(2)
[DEBUG] [1736289534.656185853]: Connection::drop(2)
[DEBUG] [1736289534.745119443]: Received update for topic [/wx250s/commands/joint_group] (0 publishers)
[DEBUG] [1736289534.745195970]: Publisher update for [/wx250s/commands/joint_group]:  already have these connections: http://128.112.36.33:45681/, 
[DEBUG] [1736289534.745233938]: Disconnecting from publisher [/wx250s_robot_manipulation] of topic [/wx250s/commands/joint_group] at [http://128.112.36.33:45681/]
[DEBUG] [1736289534.745259275]: Connection::drop(2)
[DEBUG] [1736289534.745299104]: Connection::drop(2)
[DEBUG] [1736289534.745331890]: Connection::drop(2)
[DEBUG] [1736289534.746541045]: Received update for topic [/wx250s/commands/joint_single] (0 publishers)
[DEBUG] [1736289534.746588650]: Publisher update for [/wx250s/commands/joint_single]:  already have these connections: http://128.112.36.33:45681/, 
[DEBUG] [1736289534.746613270]: Disconnecting from publisher [/wx250s_robot_manipulation] of topic [/wx250s/commands/joint_single] at [http://128.112.36.33:45681/]
[DEBUG] [1736289534.746643029]: Connection::drop(2)
[DEBUG] [1736289534.746678766]: Connection::drop(2)
[DEBUG] [1736289534.746703121]: Connection::drop(2)
[DEBUG] [1736289534.747844568]: Received update for topic [/wx250s/commands/joint_trajectory] (0 publishers)
[DEBUG] [1736289534.747888756]: Publisher update for [/wx250s/commands/joint_trajectory]:  already have these connections: http://128.112.36.33:45681/, 
[DEBUG] [1736289534.747922333]: Disconnecting from publisher [/wx250s_robot_manipulation] of topic [/wx250s/commands/joint_trajectory] at [http://128.112.36.33:45681/]
[DEBUG] [1736289534.747951734]: Connection::drop(2)
[DEBUG] [1736289534.747983963]: Connection::drop(2)
[DEBUG] [1736289534.748013840]: Connection::drop(2)
```

abadithela, 2025-01-09:
Hi, 

We're still facing this issue about 1 in 3 experiments. Is there any way to diagnose this issue? I've plotted the joint states over the course of operation (the relevant part of the plot is after time step 100).

[image]

Best,
Apurva

closed 2025-03-07 by abadithela