Interbotix/interbotix_ros_manipulators #241: [Bug]: Joint Pub GUI not starting

abadithela, 2025-01-08:
### What happened?

Hi,

I'm running the widowx 250s on ROS 1. I'm trying to debug a few failures during robot operation. For this, I'm trying to monitor the joint states, and I am trying to launch ROS using the following command (as instructed [here](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros1_packages/arm_descriptions.html#usage)):

`roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s use_joint_pub_gui:=true`

However, this does not seem to open any new window showing the joint states. There seems to be no change between running the above command and running the following command:

`roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s`

### Robot Model

wx250s

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 1 Noetic

abadithela, 2025-01-08:
OK, thanks. I changed the parameter to true in the launch file you have stated. What GUI should I expect to see? 

When I startup the robot, I still get:

```(base) iromlab@treacle:~$ roslaunch interbotix_xsarm_control xsarm_control.launch robot_model:=wx250s 
... logging to /home/iromlab/.ros/log/277ce116-cde6-11ef-bb2e-f73a6cacbca2/roslaunch-treacle-181841.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://128.112.36.33:43507/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.17.0
 * /wx250s/robot_description: <?xml version="1....
 * /wx250s/xs_sdk/load_configs: False
 * /wx250s/xs_sdk/mode_configs: /home/iromlab/int...
 * /wx250s/xs_sdk/motor_configs: /home/iromlab/int...

NODES
  /wx250s/
    joint_state_publisher_gui (joint_state_publisher_gui/joint_state_publisher_gui)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    xs_sdk (interbotix_xs_sdk/xs_sdk)

auto-starting new master
process[master]: started with pid [181889]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 277ce116-cde6-11ef-bb2e-f73a6cacbca2
process[rosout-1]: started with pid [181918]
started core service [/rosout]
process[wx250s/joint_state_publisher_gui-2]: started with pid [181925]
process[wx250s/robot_state_publisher-3]: started with pid [181927]
process[wx250s/rviz-4]: started with pid [181928]
process[wx250s/xs_sdk-5]: started with pid [181929]
[ INFO] [1736357386.831608000]: [xs_sdk] Loaded mode configs from '/home/iromlab/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_control/config/modes.yaml'.
[ INFO] [1736357386.833026345]: [xs_sdk] Loaded motor configs from '/home/iromlab/interbotix_ws/src/interbotix_ros_manipulators/interbotix_ros_xsarms/interbotix_xsarm_control/config/wx250s.yaml'.
[ INFO] [1736357386.833819273]: [xs_sdk] Pinging all motors specified in the motor_config file. (Attempt 1/3)
[ INFO] [1736357386.868309256]: [xs_sdk]	Found DYNAMIXEL ID:  8, Model: 'XL430-W250', Joint Name: 'wrist_rotate'.
[ INFO] [1736357386.914279259]: [xs_sdk]	Found DYNAMIXEL ID:  7, Model: 'XM430-W350', Joint Name: 'wrist_angle'.
[ INFO] [1736357386.960263836]: [xs_sdk]	Found DYNAMIXEL ID:  5, Model: 'XM430-W350', Joint Name: 'elbow_shadow'.
[ INFO] [1736357387.006249791]: [xs_sdk]	Found DYNAMIXEL ID:  4, Model: 'XM430-W350', Joint Name: 'elbow'.
[ INFO] [1736357387.052273230]: [xs_sdk]	Found DYNAMIXEL ID:  6, Model: 'XM430-W350', Joint Name: 'forearm_roll'.
[ INFO] [1736357387.098311347]: [xs_sdk]	Found DYNAMIXEL ID:  2, Model: 'XM430-W350', Joint Name: 'shoulder'.
[ INFO] [1736357387.144239286]: [xs_sdk]	Found DYNAMIXEL ID:  9, Model: 'XL430-W250', Joint Name: 'gripper'.
[ INFO] [1736357387.190288061]: [xs_sdk]	Found DYNAMIXEL ID:  3, Model: 'XM430-W350', Joint Name: 'shoulder_shadow'.
[ INFO] [1736357387.236246308]: [xs_sdk]	Found DYNAMIXEL ID:  1, Model: 'XM430-W350', Joint Name: 'waist'.
[ INFO] [1736357387.248732478]: [xs_sdk] Skipping writing startup register values to EEPROM.
[ INFO] [1736357387.702310433]: [xs_sdk] The operating mode for the 'arm' group was changed to position.
[ INFO] [1736357387.736292048]: [xs_sdk] The operating mode for the 'gripper' joint was changed to pwm.
[ INFO] [1736357387.951884789]: [xs_sdk] Interbotix 'xs_sdk' node is up!
```

And, when I kill ROS, I get a message about the joint_state_gui:

```
^C[wx250s/xs_sdk-5] killing on exit
[wx250s/rviz-4] killing on exit
[wx250s/robot_state_publisher-3] killing on exit
[wx250s/joint_state_publisher_gui-2] killing on exit

[wx250s/joint_state_publisher_gui-2] escalating to SIGTERM
[wx250s/joint_state_publisher_gui-2] escalating to SIGKILL
[rosout-1] killing on exit
[master] killing on exit
Shutdown errors:
 * process[wx250s/joint_state_publisher_gui-2, pid 181925]: required SIGKILL. May still be running.
shutting down processing monitor...
... shutting down processing monitor complete
done
```

lukeschmitt-tr, 2025-01-08:
You should expect to see the gui shown in the first picture in [this guide](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros1/quickstart.html#ros-1-quickstart-guide).

Note that the xs_sdk and the joint pub gui are not meant to be used at the same time since they publish to the same topic - the xs_sdk publishes real states and the gui publishes fake states. The gui is meant to be a way to explore possible configurations of a robot using a virtual model.

abadithela, 2025-01-08:
Got it, thanks! Is there a gui to plot joint states while xs-sdk is in operation? Right now, I am printing joint states to terminal but it would be useful to have a visual plot. I tried rqt_plot /wx250s/joint_states, but that does not plot anything either. Where is the JointState Msg defined?

lukeschmitt-tr, 2025-01-08:
> /wx250s/joint_states

Try something like `/wx250s/joint_states.position[0]` to get the position of the first joint.

> Where is the JointState Msg defined?

You can see the message definition [here](https://docs.ros.org/en/latest/api/sensor_msgs/html/msg/JointState.html).

abadithela, 2025-01-08:
Thank you. The following command worked: `rqt_plot /wx250s/joint_states/position[0]`. I then tried plotting all relevant joint states during the robot operation:

`rqt_plot /wx250s/joint_states/position[0]:position[1]:position[2]:position[3]:position[4]:position[5]:position[6]`

But, I do not see any significant change in the joint states though the gripper is moving across the table. 

[image]

lukeschmitt-tr, 2025-01-08:
It looks like the autoscroll option is not checked. Because of that, you're only seeing about the first second of position info.

abadithela, 2025-01-08:
got it!

closed 2025-01-08 by abadithela