Interbotix/interbotix_ros_manipulators #242: [Bug]: `u2d2` not recognized

kausiksivakumar, 2025-01-17:
### What happened?

When I plug the usb port to my computer and run `ls /dev | grep ttyDXL`, I get nothing. 
`lsusb` does show a device from `Future Technology Devices International`

Running `ls /dev | grep ttyUSB` gives me `ttyUSB0`

I do not know what to do here, would appreciate any help 

### Robot Model

px100

### Operating System

Ubuntu 22.04

### ROS Distro

ROS 2 Humble

### Relevant log output

```shell

```

kausiksivakumar, 2025-01-17:
If I move forward to the next steps, and I run `ros2 launch interbotix_xsarm_descriptions xsarm_description.launch.py robot_model:=wx200 use_joint_pub_gui:=true`, I do not see the joint state publisher at all

DavidHandelman1, 2025-02-06:
I had the same problem because I was trying to run the px100 arm on my Windows laptop, and although I used a Ubuntu virtual machine, the port wouldn't map correctly. Same problem of ttyUSB0. Once I used a Raspberry Pi with ROS2 Humble and Jammy Jellyfish OS, I was able to get the arm up and running.

lukeschmitt-tr, 2025-02-11:
@kausiksivakumar How did you go about installing the interbotix packages? Did you use our installation script [as documented in our ROS 2 guide](https://docs.trossenrobotics.com/interbotix_xsarms_docs/ros_interface/ros2/software_setup.html#amd64-architecture)? If so, udev rules should have been installed at `/etc/udev/rules.d/99-interbotix-udev.rules`. Please check for this file.

DavidHandelman1, 2025-02-11:
@lukeschmitt-tr Sorry if this is a dumb question, but is there any possible way to get the px100 arm running on a Windows laptop, maybe through using Ubuntu VM? I tried it but couldn't get it to work.

lukeschmitt-tr, 2025-02-12:
@DavidHandelman1 You may be able to get the arm working using WSL or some other virtualization-based method, but we do not support that at this time.